ssilverman/TeensyDMX #1: Serial2-6 for recieve on Teensy 3.6

ScholliYT, 2017-07-29:
Seems like Serial2 and Serial4 (so far as I tested) are not working for recieve on my Teensy 3.6. I connected a DMX Singal to Serial1 (Pin0) and everything works fine. Connecting it to Serial2 and Serial4 I cant recieve any values. I'm using the Flasher.ino provided in your examples. 

Would be nice if you can fix that :) 
Or is this just a problem on my side?

ssilverman, 2017-08-02:
I haven't yet added support for those serial ports.

ssilverman, 2017-09-08:
I made a new branch, "AddSerials4to6". Try that out and see if it works for you. If it does, I'll commit it to the master branch.

ScholliYT, 2017-09-12:
Ok I'll give it a try

ssilverman, 2017-09-30:
I just made a new version and tag, v2.2.0. I'll just mark this as closed and if there's any problems with the new code, I'll make a new issue.

closed 2017-09-30 by ssilverman