ssilverman/TeensyDMX #3: TeensyDMX.h:20:19: fatal error: cstdint: No such file or directory

Sorontik, 2019-01-31:
Hello,

this library sounds awesome, i've been thinkin about doing some DMX projects on my Arduinos for Months.
Unfortunately, trying to compile a Sketch using the lib yields the following error:

<correct path>libraries\TeensyDMX\src/TeensyDMX.h:20:19: fatal error: cstdint: No such file or directory

Full Sketch:

`
#include <TeensyDMX.h>

namespace dmx = ::quindesign::teensydmx;

// send / Drivers
#define DE 2
// receive
#define RE 3

dmx:Sender sender{Serial};

void setup() {
  // put your setup code here, to run once:

}
void loop() {
  // put your main code here, to run repeatedly:
}
`

Library Version: 3.0.0-beta installed from Arduino Library Manager
Arduino Version: 1.8.7

ssilverman, 2019-01-31:
Hi, @Sorontik. Thanks for your interest in the library. Some points:

1. This library is designed for the Teensy and not a generic Arduino board.
2. Only serial ports that are instances of `HardwareSerial` can be used. `Serial`, at least on the Teensy, is not. `Serial1` and so on are.
3. `cstdint` is a C++ header; this library expects to be compiled with a C++ compiler. I know that the Teensy firmware uses a C++ compiler, but maybe whatever board support the IDE has for your board isn't using one?

Which board are you using? Also, have you seen the BasicSend example?

Sorontik, 2019-01-31:
Thanks for the fast reply :D

I'm trying to use it on an Arduino Nano (similar to Teensy 2.0 with an AtMega328 instead of 32U4), where `Serial` is the (only available) `HardwareSerial`.

I'm not entirely sure about the C++ support, but other libraries defined in '.cpp'-files work just fine and i think they are using non-C features.

Just looked it up:
Arduino uses `avr-gcc 4.8.1` with languages c and c++ enabled, but libraries libssp and libada disabled

3 and 4 are totally correct typos (I'm from Germany where a 'q' is automatically followed by an 'u' in normal language)

ssilverman, 2019-02-01:
TeensyDMX needs to run on an ARM Cortex-M4 because that’s how I’ve written all the serial code; it directly accesses the hardware. It doesn’t use the Arduino API. That’s why it has to run on a Teensy. (Note that I haven’t tested this on other ARM-based systems, but I’ve tested this on the Teensy.)

The library appears as available to you, unfortunately, because the Arduino IDE doesn’t have a working way to determine the correct architecture/board for a library.

In summary, this won’t work on your board. I recommend trying out a Teensy anyway, if you’re so inclined; they’re fast and powerful.

And ‘q’ not followed by a ‘u’ is strange in English too. :)  ‘qin’ is from an old username of mine.

Sorontik, 2019-02-01:
I see...

Thanks for your support anyway :D

closed 2019-02-01 by Sorontik