ssilverman/TeensyDMX #5: Any plans for a DMX USB Pro Widget implementation?

BarakChamo, 2019-09-29:
Hi Shawn,

Thanks for your great work on this library, it's awesome to have a functional interface for working with DMX on the Teensies.

I saw you briefly mentioned that the library should be in a state where the widget API could be implemented on top and was wondering if you have a reference implementation or plans to integrate such functionality into `TeensyDMX`. Being able to do so would make it perfect for use in headless setups with DMX recorders etc. and without having to user ArtNet and additional expensive Ethernet breakout.

I understand the DMX over USB widget API is essentially a [serial signaling spec](https://dol2kh495zr52.cloudfront.net/pdf/misc/dmx_usb_pro_api_spec.pdf) but I'm not entirely sure where to begin in implementing it against software like MadMapper / TouchDesigner or ay other PC client.

Any help / pointers / reference code would be appreciated!

ssilverman, 2019-09-29:
I’ve implemented a working one a while ago, I’m still deciding if and how I want to release it. Send me an email and we can chat further?

Side question: Did your needs include RDM? Doing an RDM controller involves detecting BREAK-less packets and, while I’ve got a test version of that going too, I’m not 100% happy with how I’ve implemented it. One of those features that I’m waiting until it ‘feels right’.

BarakChamo, 2019-09-30:
Hi Shawn, I couldn't find your email, would you mind pinging me at barak.chamo@gmail.com?

My needs mainly have to do with the devices being detected automatically as a DMX-over-USB device so no specific remote control features. I imagine it's mostly the handshake and detection part that is implemented?

ssilverman, 2019-10-07:
I provided some non-commercial code that works with OLA. It _should_ work with TouchDesigner and MadMapper too, but it hasn’t been tested yet. Closing for now.

closed 2019-10-07 by ssilverman