ssilverman/TeensyDMX #8: Returning all Zeroes, Packet time negative (-44)

imwhocodes, 2020-07-16:
Hi

I'm using your library, but I'm having problems reading the data.

Using `teensydmx::Receiver::readPacket` i'm trying to read **2 channels**, sporadically both are setted to **0** even if the function has return value **2** (channel readed) , when this happen `Receiver::PacketStats::packetTime` (obtained passing it as forth function parameter) has value **-44**

This behavior is the same with both and without `teensydmx::Receiver::setRXWatchPin` configured

I'm also running 2 analogWrite, both using **FTM1** as timer

I'm running on a teensy3.2, TeensyDMX4.0.0, compiled on Platformio (Teensyduino 1.53)

Any ideas? Thanks

ssilverman, 2020-09-01:
I haven’t forgotten about this issue. Sorry for the delay. I’m having to work on some other things at the moment, but I will be getting to this.

imwhocodes, 2020-09-07:
Actually I'm sorry the root cause was not on your side.

After inspecting the signal with an oscilloscope I realised the the bits stream was not continuous having random gaps in the DMX packet

The real problem was the converter ["OPEN DMX USB"](https://www.enttec.co.uk/en/product/controls/dmx-usb-interfaces/open-dmx-usb/) having a really poor timing and no packet buffering

Using a [better](https://www.enttec.com/product/controls/dmx-usb-interfaces/dmx-usb-pro-interface/) (packet buffered) converter I didn't have any problems whatsoever

(btw I don't have any affiliation with the linked products, only sharing an experience)

Thank you for your library!

ssilverman, 2020-09-07:
@imwhocodes I'm glad you found the issue.

Some notes (at the very least for future viewers of this issue report):
1. It's possible for values to change each time they're being read. I suspect the transmitter is actually sending those **0** values, especially since the `readPacket` call is returning the same size.
2. `packetTime` can't be negative because its type is `uint32_t`.
3. For the Rx watch pin, is it actually physically connected? It won't do much to affect the read results, other than to determine more accurate timing checks within the packet. Specifically, it helps the library determine exactly where the BREAK transitions to the MAB (mark after break).

closed 2020-09-08 by ssilverman