ssilverman/TeensyDMX #9: Teensy 4.1 - No Serial8

s4CD, 2020-08-03:
Using the BasicSend.ino example works on the 4.1 with Serial1-Serial7, but not on Serial8. It will not fail to compile or during runtime, but there is no output. I tried to dig through and did notice the switch for Serial8 references the wrong IC when checking defines (IMXRT1052 instead of IMXRT1062 on the 4.0 and 4.1, and I believe the 4.1 is the only model with Serial8 available)
I tried changing that but resulted in a slew of errors. If you do not have a 4.1 available to test, I can try any updates you make to confirm. I'm just not sure where to start to try to find the issue myself.

On another note, I'm also using NativeEthernet on this board and teensyDMX seems to kill the linkStatus functionality due to link_callback not being received, yet works fine without this lib initialized.. don't expect a definitive answer but if you have ANY idea what in here may conflict with that so I could dig a bit I would love to hear.. been fighting that one for a while..

ssilverman, 2020-09-01:
I’ll have a look when I have a chance. Thank you for reporting this.

ssilverman, 2020-09-07:
Hi, @s4CD. Thank you for your patience. I've created a https://github.com/ssilverman/TeensyDMX/tree/teensy41-serial8 branch to add support for Serial8 on Teensy 4.1. If you don't mind, try this out on your Teensy 4.1 and let me know if the update works.

s4CD, 2020-09-08:
Awesome! Thank you! I will try to test it out tonight.

Chris Dalio
Product Design Engineer & Tech Lead, S4 Lights
m:
e:
w:

 1.609.529.4483<tel:1.609.529.4483>
 chris@s4lights.com <mailto:chris@s4lights.com%20>
 s4lights.com<http://s4lights.com/>

ssilverman, 2020-09-08:
For posterity, the checks for IMXRT1052 are there because that's the chip used in the pre-release boards and I want the library to work with those too.

ssilverman, 2020-09-08:
@s4CD For the NativeEthernet conflict, can you file another issue for that? We can track the conflict there instead.

s4CD, 2020-09-08:
That makes sense. And yes, I will add another issue for that. I threw it in as an afterthought because I’m not 100% sure the issue is in your code, but it definitely only happens when you library is loaded.

Chris Dalio
Product Design Engineer & Tech Lead, S4 Lights
m:
e:
w:

 1.609.529.4483<tel:1.609.529.4483>
 chris@s4lights.com <mailto:chris@s4lights.com%20>
 s4lights.com<http://s4lights.com/>

ssilverman, 2020-09-08:
There’s quite possibly a legitimate conflict with hot interrupts are done, timers are used, etc.

s4CD, 2020-09-09:
I tried it out and it seems to work great! I will let it run overnight and see if anything strange pops up, but so far so good! I will give another update in the morning.

ssilverman, 2020-09-09:
Thanks for verifying, @s4CD!

Side point: So I drafted a new beta release, 4.1.0-beta. Do you feel like testing that out? There are some major changes to simplify the code and I'd love someone external to try it out.

s4CD, 2020-09-09:
Absolutely! Anything specific I should check? Or just that is still functioning as before?

Just to note, I am using this lib with equipment that does not support RDM, so I'm only sending output, not reading returned data.

s4CD, 2020-09-10:
I've been running things all day without any issue. I would say that this issue is solved from the limits of what I have to test with.

closed 2020-09-12 by ssilverman

ssilverman, 2023-01-16:
Note: I made my own Ethernet library that seems to work well with TeensyDMX:
https://github.com/ssilverman/QNEthernet

It’s very stable and I’ve used it now in a few in-the-field projects.