ssilverman/TeensyDMX #13: Teensy4.1 multi DMX Sender.

heoseongdae, 2022-04-23:
How many serial ports can be used at the same time?
I want to use all 8 serial ports on Teensy4.1 to send out DMX signals.
 By modifying the sample code and adding the number of serial ports one by one, there is no problem at around 3, but beyond that, the processor dies.

Test Code..

#include <TeensyDMX.h>

namespace teensydmx = ::qindesign::teensydmx;

constexpr uint8_t kTXPin = 17;

teensydmx::Sender dmxTx{Serial1};
teensydmx::Sender dmxTx2{Serial2};
teensydmx::Sender dmxTx3{Serial3};
teensydmx::Sender dmxTx4{Serial4};
teensydmx::Sender dmxTx5{Serial5};
teensydmx::Sender dmxTx6{Serial6};
teensydmx::Sender dmxTx7{Serial7};
teensydmx::Sender dmxTx8{Serial8};

// Data for 3 channels.
uint8_t data[3]{0x44, 0x88, 0xcc};

void setup() {
   Serial.begin(128000);
   while (!Serial && millis() < 4000) {
  }
  // Turn on the LED, for indicating activity
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWriteFast(LED_BUILTIN, HIGH);

  // Set the pin that enables the transmitter; may not be needed
  pinMode(kTXPin, OUTPUT);
  digitalWriteFast(kTXPin, HIGH);

  dmxTx.set(1, 128);
'
  dmxTx.set(10, data, 3);

  dmxTx.begin();
  dmxTx2.begin();
  dmxTx3.begin();
//  dmxTx4.begin();
//  dmxTx5.begin();
//  dmxTx6.begin();
//  dmxTx7.begin();
//  dmxTx8.begin();
}

void loop() {
  Serial.print( dmxTx.packetSize());
  Serial.print(" ");
  Serial.print( dmxTx2.packetSize());
  Serial.print(" ");
  Serial.print( dmxTx3.packetSize());
  Serial.print(" ");
  Serial.print( dmxTx4.packetSize());
  Serial.print(" ");
  Serial.print( dmxTx5.packetSize());
  Serial.print(" ");
  Serial.print( dmxTx6.packetSize());
  Serial.print(" ");
  Serial.print( dmxTx7.packetSize());
  Serial.print(" ");
  Serial.println( dmxTx8.packetSize());
  //Serial.println( dmxTx.refreshRate());
  dmxTx.set(10,data,3);
  dmxTx2.set(10,data,3);
  dmxTx.setPacketSize(data[0]);
  dmxTx2.setPacketSize(data[0]+10);
  dmxTx3.setPacketSize(data[0]+20);
//  dmxTx4.setPacketSize(data[0]+30);
//  dmxTx5.setPacketSize(data[0]+40);
//  dmxTx6.setPacketSize(data[0]+50);
 // dmxTx7.setPacketSize(data[0]+60);
 // dmxTx8.setPacketSize(data[0]+70);
 data[0]++;
  
  delay(100);
}

ssilverman, 2022-04-23:
I copied the code, uncommented transmitters 4-8, and the program seems to work fine. I tried with TeensyDMX v4.2.0 and the latest v4.3.0-snapshot from GitHub. What sort of hardware do you have connected?

heoseongdae, 2022-04-23:
I tried with TeensyDMX v4.2.0 and the latest v4.3.0-snapshot from GitHub. Both versions have the same thing.
After one output from the loop, the processor is unresponsive.
The hardware I used is Teensy 4.1 and I didn't connect any other hardware to the Teensy board just to test the code.
I'm ready to test out a new Teensy board.

heoseongdae, 2022-04-26:
Tested with a new board.
It runs fine with no problems. Thank you for your response.
I suspect it may have been rotten with a Chinese copyboard.
I've attached a photo.
All of the markings on the processor are identical, but the marking state is very different.
And the C6 and C3 markings on the back are odd too.
Thanks for you help.

[image]
[image]