ssilverman/TeensyDMX #15: USBWidgetPro Example Correction

welshcoder, 2022-06-20:
Hi,

First of all, thank you for this great library!

I've been using this for a couple of projects of mine, and it works very well. One of the projects is to upgrade the Velleman VM116/K8062 USB Controller DMX Interface by removing the original PIC and replacing it with a TeensyLC running TeensyDMX in send mode. I essentially used the USBWidgetPro example pretty much as is, and it works well with FreeStyler and the fixtures I tested it on.

However, when I tried to use it controlling another one of my projects, I was getting quite a lot of glitches coming through. To cut a long story short, it turned out that the problem was that the TeensyLC was pausing briefly to process data received from the USB port.

The issue is this block of code in `handleMessage` (line 508):
```
if (len != dmxTx.packetSize()) {
   __disable_irq();
   dmxTx.setPacketSize(len);
   dmxTx.set(0, msg.data, len);
   __enable_irq();
   } else {
     dmxTx.set(0, msg.data, len);
}
```
The data can be set whilst a transmission is taking place, causing a brief delay in the DMX data stream and glitching the receiver. I propose changing this block to:

```
if (len != dmxTx.packetSize()) {
  dmxTx.pause();
  while(dmxTx.isTransmitting()) { yield(); }
  dmxTx.setPacketSize(len);
  dmxTx.set(0, msg.data, len);
  dmxTx.resume();
} else {
  dmxTx.pause();
  while(dmxTx.isTransmitting()) { yield(); }
  dmxTx.set(0, msg.data, len);
  dmxTx.resume();
}
```
After this change, the glitches disappeared and it all works perfectly!

I hope it helps!

ssilverman, 2022-06-20:
Thanks for this report. Question: why would slightly longer delays in the DMX data stream affect the receiver? To me, that would mean a receiver that doesn’t conform to the DMX spec. Do you happen to have oscilloscope or logic analyzer captures of the DMX output that causes a receiver glitch?

welshcoder, 2022-06-21:
Thanks for getting back to me.

The receiver in this case is a Teensy 4.1 with an isolated RS485 interface to the DMX connector. I'm using the TeensyDMX library to process the incoming DMX data on the 4.1. I tried using the Rx monitor option for the timing, but it didn't seem to make much difference. What I was seeing on the receiver using the readPacket function was a series of zeros when it was glitching, followed by the correct (non-zero) data. When I connected a commercial stand-alone DMX controller to the receiver, it worked perfectly - no glitching whatsoever.

I do have an oscilloscope, and I was using it to monitor the USB data lines and the DMX output. Unfortunately, I didn't take screen captures! If I have time later on, I'll hook it up again and capture the issue using the original code. In a nutshell, what I was seeing is shortly after the burst of USB data, the output of the DMX would freeze mid-frame for a period of about 50-100 us (from memory - need to confirm) before continuing. With the revised code, that delay is added to the end of the frame and does not affect the frame timing.

Does that help?

ssilverman, 2022-06-21:
Some questions:
1. Which version of Teensyduino?
2. Which version of TeensyDMX?
3. Can you show me what your receiver code looks like?

welshcoder, 2022-06-21:
Sure, no problem.

1. Version 1.156
2. Both are using the version available through PlatformIO library manager: version 4.3.0-snapshot?
3. Here is the complete DMX setup and loop sections in my code. It should be fairly straightforward: `setupDMXSubSystem` is run during the `setup()` process, and `loopDMXSubSystem` runs during the `loop()` process. `DMXValues` is the array containing the relevant DMX values to the receiver starting from `mSettings.DMXStartAddress`.

```
#ifndef _DMX_SUB_SYSTEM_H_
#define _DMX_SUB_SYSTEM_H_

#include <TeensyDMX.h>

using namespace ::qindesign::teensydmx;

Receiver DMXRx{Serial8};

enum DMXChannelDef : uint8_t {
    DMX_CH_MODE = 0,
    DMX_CH_RED1,
    DMX_CH_GREEN1,
    DMX_CH_BLUE1,
    DMX_CH_STROBE,
    DMX_CH_RED2,
    DMX_CH_GREEN2,
    DMX_CH_BLUE2,
    DMX_CH_DIMMER,
    DMX_CH_LENGTH
};

enum DMXModeDef : uint8_t {
    DMX_MODE_FULL_SCREEN = 0x00,
    DMX_MODE_HALF_SCREEN = 0x10,
    DMX_MODE_QURT_SCREEN = 0x20,
    DMX_MODE_ANIM_MODULE = 0x30,
    DMX_MODE_FRCT_MODULE = 0x40,
    DMX_MODE_VIDO_MODULE = 0x50,
};

array<uint8_t,DMX_CH_LENGTH> DMXValues;

void enableDMXTx(bool en=true)
{
    DMXRx.setTXEnabled(en);
    digitalWriteFast(DMX_TX_EN_PIN,en?HIGH:LOW);
}

void setupDMXSubSystem()
{
    pinMode(DMX_TX_EN_PIN,OUTPUT);
    enableDMXTx(false);

    DMXRx.begin();
}

bool loopDMXSubSystem()
{
    uint8_t packetBuf[DMX_CH_LENGTH];
    int read=DMXRx.readPacket(packetBuf,mSettings.DMXStartAddress,DMX_CH_LENGTH);

    if(read==DMX_CH_LENGTH)
    {
        if(memcmp(packetBuf,DMXValues.data(),DMX_CH_LENGTH)!=0)
        {
            memcpy(DMXValues.begin(),packetBuf,DMX_CH_LENGTH);

            return true;
        }
    }

    return false;
}

#endif
```

4. I don't think it was anywhere in particular in the frame. If I remember correctly, it was pretty consistent from the timing of the USB data burst, regardless of what the transmitted DMX data stream was doing at the time. I would need to confirm that though. It was definitely occurring at seemingly random times in the frame, as I could see the glitch jumping around when I zoomed out to see the whole frame on the scope.

I believe the "else" block of the original code that doesn't have the interrupt control was being executed, so I can see how this may come about.

ssilverman, 2022-06-21:
Do either of the following changes still cause a glitch (I want to see which of the cases gets executed --- it's the change-one-thing-at-a-time approach):

First:
```c++
if (len != dmxTx.packetSize()) {
  __disable_irq();
   dmxTx.setPacketSize(len);
   dmxTx.set(0, msg.data, len);
   __enable_irq();
} else {
  dmxTx.pause();
  while(dmxTx.isTransmitting()) { yield(); }
  dmxTx.set(0, msg.data, len);
  dmxTx.resume();
}
```

Second:
```c++
if (len != dmxTx.packetSize()) {
  dmxTx.pause();
  while(dmxTx.isTransmitting()) { yield(); }
  dmxTx.setPacketSize(len);
  dmxTx.set(0, msg.data, len);
  dmxTx.resume();
} else {
  dmxTx.set(0, msg.data, len);
}
```

More questions:
1. Why are you enabling TX on the receiver? Are you trying to do RDM or something? (`DMXRx.setTXEnabled(en)`)

welshcoder, 2022-06-21:
Hopefully, I'll get a chance to try out the code tonight and let you know how it goes.

In the meantime, the answer to your questions are:
1. I wrote the function with the intention of implementing RDM in the future, but I set it to false in `setupDMXSubSystem`, and doesn't change from there. Therefore, the receiver is always in receive-only mode.
2. It manifests itself by the block of data I obtain from `readPacket` being all zeros on the receiver, and then the receiver is acting on that data as if it's valid. Since the TeensyLC is constantly writing DMX data, the receiver then obtains the correct data in the next frame for the channels I need (i.e. the DMX stream to that point has no glitches). I've confirmed that the `mSettings.DMXStartAddress` is unchanged for all the calls made to that function. This behaviour appeared to be random too. When I implement my proposed code changes, this behaviour disappeared completely and no glitching occurs.

welshcoder, 2022-06-21:
Ok, got some captures from the scope with and without the glitching. In all the captures, the yellow trace is the DMX stream, and the blue(ish) trace is the USB data line. I used the second code you asked me to try above for the glitching version, and my proposed code changes for the no-glitch version.

**Glitching 1**
[image]
In this image, you can see the data burst on the USB line, followed about 1.5 ms later by the DMX stream freezing slightly (stays high). All the DMX values in that section are meant to be 0, hence should be a regular pattern.

**Glitching 2**
[image]
Here's another example showing the same thing but at a different time. The first few channels have non-zero values in them, hence you see some data near the beginning of the frame.

**Glitching 3**
[image]
This is showing the same as above. However, note that the USB burst is at a different position relative to the DMX frame start, but the glitch in the DMX stream occurs more-or-less at the same interval after the USB data burst has ended.

**No glitch 1**
[image]
This is the situation after I've implemented my changes. You'll notice there is no glitching after the USB data bursts. The frame start after the first USB data burst is slightly longer than the next one, as the processing of the incoming data has been delayed until the end of the frame.

**No glitch 2**
[image]
Same as the previous one, but zoomed in to show that the glitch doesn't appear in the DMX stream after the code change.

ssilverman, 2022-06-23:
I just noticed your edit on one of your comments saying the glitch happens in my second test code block (i.e. the “else” block). Can you try this block of code:

```c++
if (len != dmxTx.packetSize()) {
  __disable_irq();
  dmxTx.setPacketSize(len);
  dmxTx.set(0, msg.data, len);
  __enable_irq();
} else {
  __disable_irq();
  dmxTx.set(0, msg.data, len);
  __enable_irq();
}
```

welshcoder, 2022-06-23:
Sorry for the delay in getting back to you, it's been a rather busy couple of days for me.

There are a few things to cover here, so I'll try to do this as methodically as possible!

Regarding the atomic interrupt blocks, I used this code to try it out:
```
      if (len != dmxTx.packetSize()) {
        ATOMIC_BLOCK(ATOMIC_FORCEON)
        {
            dmxTx.setPacketSize(len);
            dmxTx.set(0, msg.data, len);
        }
      } else {
        ATOMIC_BLOCK(ATOMIC_FORCEON)
        {
            dmxTx.set(0, msg.data, len);
        }
      }
```
This code causes the glitching too.

In reference to https://github.com/ssilverman/TeensyDMX/issues/15#issuecomment-1163988662, this code causes the glitching as before.

One thing I did notice, which I don't think I noticed before, is that the Teensy 4.1 in the receiver crashed after a short period of time. Now, that could be entirely due to an unrelated problem, but reverting to my version of the TeensyLC code that stops the glitching doesn't cause this and the T4.1 is pretty stable. The T4.1 is doing a lot of work in the background (it's running SmartMatrix library and driving a 128 x 128 RGB matrix), so it could have something to do with that. Although I have already spent quite a lot of time on the receiver code before raising this issue on here, I've had a closer look on what's going on.

I did the test of using `readPacket` to gather all the data from the frame, and the value was not consistent on each frame with both the glitching and non-glitching code, see below (ignore the '0:' - that's just the start address I was using). 
Glitching code
```
0: 513
0: 325
0: 361
0: 172
0: 19
0: 451
0: 451
0: 513
0: 513
0: 513
0: 513
0: 193
0: 513
0: 513
0: 513
0: 513
0: 513
0: 355
0: 513
0: 397
0: 184
0: 513
0: 513
0: 513
0: 513
0: 513
0: 199
0: 513
0: 513
```
Non-glitching code
```
0: 513
0: 235
0: 250
0: 513
0: 513
0: 410
0: 55
0: 513
0: 513
0: 438
0: 508
0: 513
0: 241
0: 76
0: 513
0: 513
0: 513
0: 358
0: 260
0: 250
0: 513
0: 513
0: 315
0: 513
0: 46
0: 513
0: 513
0: 446
0: 275
0: 287
0: 268
0: 116
0: 361
0: 296
0: 513
0: 513
0: 513
0: 513
0: 513
0: 513
0: 258
0: 513
0: 513
0: 513
0: 513
0: 486
0: 259
0: 513
0: 382
0: 513
0: 513
0: 513
0: 268
0: 513
0: 513
0: 101
0: 307
0: 513
0: 371
0: 229
0: 513
0: 513
0: 513
0: 79
0: 304
0: 232
0: 161
0: 234
0: 449
0: 513
0: 247
0: 513
0: 513
0: 513
0: 513
0: 292
0: 513
0: 7
0: 513
0: 513
0: 513
0: 301
0: 76
0: 370
0: 151
0: 327
```

After that test, I went back to a bare-bones code for the T4.1 - a cut-down version of the BasicReceive example - and nothing else. Guess what? All glitching in the receiver disappeared! The original code as written in the USBWidgetPro example works without glitching too.

So, the issue is in the receiver, not the TeensyLC. I'm now suspecting that there is an interrupt clash, most probably with the SmartMatrix library. I'll need more time to investigate this further. The T4.1 crashes I mentioned above might provide a pointer for it. I'll see if I can track this down over the next few days, and keep you posted.

welshcoder, 2022-06-23:
Just some extra info on this. I've been trying to see how the TeensyDMX library works with the SmartMatrix one, and I've captured some data here: https://gist.github.com/welshcoder/c994c9ca2ba84acc634edfa1db52fa61

This will need a bit of an explanation!! I placed a `serial.print` in various points in the code to see what runs and when. For TeensyDMX, I placed an "I" at the beginning of `LPUARTReceiveHandler::irqHandler()` routine, and "E\n" just before `completePacket(RecvStates::kIdle);` in `Receiver::receiveByte()`. The "C" and the "S" take place in two interrupts in the SmartMatrix library, and you can see they trigger with a consistent regular pattern (triggered using DMA).

In the main loop of the program, I use `readPacket` to capture the whole frame. If the returned value isn't 513 or -1, then it prints the value of `read` plus values from some channels in the form of "7: 134 000 231 230 120".

On most lines, you see an "E" on the end which indicates that the DMX frame end has been reached. Since the output of `readPacket` isn't printed, we assume that all is well.

If you do a search for a colon, you find the areas that have the output of `readPacket` was printed, indicating a problem. The consistent issue I see is that in the previous line, the "I" disappears for a period of time, indicating that the receive interrupt in TeensyDMX seems to stop. I haven't quite worked out why yet, because it seems to play nicely with the other interrupts for most of the time.

Unfortunately, I haven't got time to look further into this until the weekend, but I'm happy to investigate further if there's anything in particular you'd like me to try. Also, if you have a T4.1, I can send you the code I'm using so you can try it out.

welshcoder, 2022-06-25:
I've been looking at this issue this morning, and I've concluded the issue seems to be with the `Lock` process in the Receiver class.

I was trying to figure out why the receiver interrupt was having a problem, and after placing some `Serial.print` statements around the code, I found that for most of the time, `packetRead` would execute very quickly and the interrupt routine would work on a very regular pattern as expected. However, if `packetRead` was executing when an interrupt was expected, the `Lock `causes the interrupt to not trigger. There are two effects of this that I noticed. The first is that `packetRead` seemed to take quite a bit longer to execute (using the SmartMatrix regular interrupt as a clock). The second effect I noticed is that the value of avail in the `irqHandler` on line 132 of `LPUARTReceiveHandler.cpp`:
`uint8_t avail = (port_->WATER >> 24) & 0x07;  // RXCOUNT`
...becomes 4 instead of the usual 3. Therefore, I suspect that the LPUART buffer is overflowing perhaps?

Although the glitching occurs pretty randomly, the above behaviour is rather consistent when it does glitch. If I comment out the `Lock` statement in the `packetRead` function (line 357 of `Receiver.cpp`), the glitching disappears. Clearly, there is a potential for a clash in accessing the same memory location, although in my testing this morning, I haven't seen any obvious effects of that.

ssilverman, 2022-06-27:
Just to make sure I understand: even with your fix, you still sometimes see an issue on the receiver side?

The `Lock` class uses RAII to simply disable and then re-enable the UART interrupt so that a certain action can be done atomically. What happens if bytes come in while the UART interrupt is disabled is they will be buffered. All you're seeing is the buffer holding one more byte than usual (4 instead of 3). The system will still process it. See the code below and notice that it loops over `avail`. In your case, It will loop 4 times instead of 3.

Is it possible that the SmartMatrix interrupts occur for longer than the duration necessary for buffering unprocessed bytes? I wonder what would happen if the UART interrupt priority is increased to something greater than the SmartMatrix interrupt.

Side note regarding your code:
```c++
      if (len != dmxTx.packetSize()) {
        ATOMIC_BLOCK(ATOMIC_FORCEON)
        {
            dmxTx.setPacketSize(len);
            dmxTx.set(0, msg.data, len);
        }
      } else {
        ATOMIC_BLOCK(ATOMIC_FORCEON)
        {
            dmxTx.set(0, msg.data, len);
        }
      }
```
It appears, from your statements above, that only the "else" part causes the glitching if we disable interrupts vs. pausing. In the spirit of changing only one thing at a time, let's just focus on changes to the "else" part.

welshcoder, 2022-06-28:
> Just to make sure I understand: even with your fix, you still sometimes see an issue on the receiver side?

Yes, that's right, but the difference was that I wouldn't get the DMX values briefly changing to zeros and back to the proper value. It was very odd.

>The Lock class uses RAII to simply disable and then re-enable the UART interrupt so that a certain action can be done atomically. What happens if bytes come in while the UART interrupt is disabled is they will be buffered. All you're seeing is the buffer holding one more byte than usual (4 instead of 3). The system will still process it. See the code below that and notice that it loops over avail. In your case, It will loop 4 times instead of 3.

Fair enough. I think there's more going on that I can't seem to work out (see below).

> Is it possible that the SmartMatrix interrupts occur for longer than the duration necessary for buffering unprocessed bytes? I wonder what would happen if the UART interrupt priority is increased to something greater than the SmartMatrix interrupt (or the SmartMatrix priority reduced). Do you happen to know what priority the SmartMatrix interrupt has?

This was my thought too, but it didn't seem to be. There would be several calls to the SM interrupt routine, during which `packetRead` would just sit there during the glitch. The SM interrupt is timed off DMA triggers, and it seems to be the highest interrupt priority available.

I'm chasing another, possibly entirely unrelated, glitch. I noticed that after running for about 30 mins with DMX constantly being received, the T4.1 would seemingly just freeze for about 7 - 10 seconds, and then continue as before. In fact, the glitch was consistent to virtually the same `millis()` value: 1889520 ms! After this time, `packetCount()` returns a value of 82289. This happens after I've disabled SM (i.e. it's loaded into memory, but its interrupts are not running) too. I'm slowly disabling different parts of the code to see that triggers the glitch. Again, I can't confirm if this is a problem with TeensyDMX or some other part of the code. But, getting there slowly!

ssilverman, 2022-06-29:
Thank you for diving into this with me. If it’s a problem with TeensyDMX, I’d love to fix it. It’s either that or some interaction with other libraries in certain circumstances and I’d like to identify those potential issues. Thanks for drilling down into the code to find it.

ssilverman, 2022-06-30:
Unless you’ve defined `TEENSYDMX_USE_PERIODICTIMER` for your build, the `PeriodicTimer` class isn’t used.

welshcoder, 2022-06-30:
I can't find `TEENSYDMX_USE_PERIODICTIMER` anywhere in the code (I will double-check this later on). Looking through, I have to define `USE_INTERVALTIMER` to using the `IntervalTimer` class though?

ssilverman, 2022-06-30:
Try getting the latest version of the library (via git or via a download). I’m curious if the behaviour will change. (It’s possible it won’t, but I’d like to rule that out.) I’m not sure which version you’re testing with.

ssilverman, 2022-07-01:
Thanks for finding that. I’ll have a closer look at that commit.

welshcoder, 2022-07-15:
Hi,

I just thought I'd give you an update on this. In a nutshell, I've struggled to get the SmartMatrix and TeensyDMX behaving well together. It may be that the screen resolution of 128 x 128, and the interrupt process to service the screen, is just too much for both libraries to work together (this is on top of the image processing routines to generate an image). One thing I do note is that the receive ISR in TeensyDMX seems to be processing quite a lot (relative to other ISRs), so that might be an area that can be improved perhaps?

As an aside, I did hack together a T4.1 version of the LXTeensy4DMX library (https://github.com/claudeheintz/LXTeensy4DMX_Library) just for the receiver, and I've managed to get stable DMX control of the display with it (no glitching at all), under the same conditions. Perhaps it could be worth seeing if there's something in that library that could make this library work better under heavier CPU loads?

ssilverman, 2022-07-15:
Just to make sure I’ve got it right: you’re seeing issues with TeensyDMX and SmartMatrix on a Teensy 4.1?

welshcoder, 2022-07-17:
Yes, that's right. Changing the priorities (SM was lower priority than TeensyDMX as it happens) didn't really help. When I made SM higher priority, the display worked fine, but virtually every single packet would be bad. Flipped it the other way, and I get the glitching of the display and some packets marked bad. By marked bad, I mean that the `receivedBadBreak` would be called.

Re: code - sure, no problem. I've put it here: https://gist.github.com/welshcoder/dc75fe40ff30a641028204670b0fc860. I've removed a lot of the commented out test code so that it's clearer as to what's going on. I think most of it is pretty self-explanatory, but let me know if you need anything else.

ssilverman, 2022-08-23:
Let me know how this fix works for you: https://github.com/ssilverman/TeensyDMX/tree/atomic-packet-size-and-data

See the commit log, but essentially I added a new `Sender::setPacketSizeAndData()` function.

welshcoder, 2022-09-06:
Sorry for the delay in getting back to you, and just a quick message to say that I haven't yet had an opportunity to look into this. I'll try to do it over the weekend and let you know if the fix works.

ssilverman, 2022-09-10:
That would be amazing. Thank you.

welshcoder, 2022-09-18:
Sorry once again for the delay in getting back to you on this. I've spent a bit of time tonight looking into this. I'm now pretty convinced there are two issues here that I'll explain separately.

1. I couldn't get the fix to work. I think this is the `a2ac5f0` commit problem that I mentioned earlier. I just wasn't getting any data out, no matter what I did, including using my version of the fix. That was confirmed with a scope too, and it looked like the "Not working code" in the earlier post. I think that needs to be looked at first.

2. The second problem is, IMHO, with the DMX receiver interrupt taking up too much time. This manifests itself as interrupting the SmartMatrix routines in the display, causing the display to breakdown for a second or two. I've managed to confirm this using a standalone DMX controller connected directly to the display (to remove the possibility of the DMX transmitter causing the problem), and I get the display glitching every now and again. Changing the DMX receiver to the LXTeensy4DMX library I mentioned before, and the glitching goes away. 

I hope that helps. Just to let you know, I'm going to be away for a couple of weeks, so I won't be able to test anything in that time. Otherwise, I'm happy to help you where I can.

welshcoder, 2022-10-29:
Hi,

I'm away at the moment, so I can't test anything to confirm my findings. But what I can say is that the branch doesn't work because it incorporates the `a2c5f0` commit. Therefore, I can't say whether your new code works or not. It might be perfect, but something else is amiss. From memory, I did try the `TEENSY_USE_PERIODICTIMER` define on the master branch a while ago, but I don't recall seeing any difference.

I'll be back home in a few days, so I'll be in a better position to test things. In the meantime, it might be worth finding out what's going on with that commit that seems to break things. Presumably, you're not seeing that particular problem?

ssilverman, 2022-11-05:
I'm staring at commit https://github.com/ssilverman/TeensyDMX/commit/a2ac5f02fd56a51a7b9935c6e783c5c07f35b9e6 and I just can't see why the logic would be different. A compiler difference, maybe?

welshcoder, 2022-11-09:
Ah, I know that feeling very well! I'm hoping that I'll have some time over the next week or so to look at this in more detail. I'll keep you posted. In the meantime, if there's anything you'd like me to try, let me know.

ssilverman, 2022-11-09:
Thanks again for your time with helping me solve this.

welshcoder, 2022-11-19:
Sorry for the delay in getting back to you on this. I've spent some time today looking at this, and I've found the problem in that commit that causes the problem. It's the change in lines 185 to 195 in `UARTSendHandler.cpp` that caused the problem for me. The original code is
```
#else  // No FIFO
if (sender_->inactiveBufIndex_ >= sender_->inactivePacketSize_) {
          setCompleting();
        } else {
          port_->D = sender_->inactiveBuf_[sender_->inactiveBufIndex_++];
          if (sender_->inactiveBufIndex_ >= sender_->inactivePacketSize_) {
            setCompleting();
          } else if (sender_->interSlotTime_ != 0) {
            sender_->state_ = Sender::XmitStates::kInterSlot;
            setCompleting();
          }
        }
#endif  // KINETISK
```
...and the new code is this:
```
#else  // No FIFO
        if (sender_->inactiveBufIndex_ < sender_->inactivePacketSize_) {
          port_->D = sender_->inactiveBuf_[sender_->inactiveBufIndex_++];
          if ((sender_->inactiveBufIndex_ < sender_->inactivePacketSize_) &&
              (sender_->interSlotTime_ != 0)) {
            sender_->state_ = Sender::XmitStates::kInterSlot;
          }
        }
        setCompleting();
#endif  // KINETISK
```

The problem is caused by all logical paths in the new code calls the `setCompleting()` function, whereas the old code, there is a logical path where the `setCompleting()` is not called (note the `else if(...)` statement). Reverting this code back to how it was fixes the problem. The equivalent code in `LPUARTSendHandler.cpp` might be a problem too.

ssilverman, 2022-11-19:
Thanks for that. However, I'm not seeing the code path that _doesn't_ call `setCompleting()` in the code snippets.

welshcoder, 2022-11-19:
If the third `if` statement (that's part of the `else if`) is false, then the `setCompleting()` function is not called at all in the original code.

ssilverman, 2022-11-19:
Thanks for your persistence and patience. I just pushed a fix.

welshcoder, 2022-11-19:
Ok, good news and bad news.

The fix you pushed earlier works, and I'm getting output again. The bad news is that the other fix hasn't really changed much in terms of the glitching. I've now got a video of the scope trace to illustrate what might be going on. 

First of all, here is the output with the following code:
```
if (len != dmxTx.packetSize()) {
     dmxTx.setPacketSizeAndData(len, 0, msg.data, len);
   } else {
     dmxTx.set(0, msg.data, len);
   }
```
https://user-images.githubusercontent.com/13755765/202872497-ce0115b7-33ad-4bfc-91ad-276eef86b944.mp4

You can see that there's a fair bit of flickering and jitter in the timing. I believe the random flickering you see every now and again coincides with the code above running. The glitching on the receiver seems to occur with code.

I then changed the code to the following and got this video:
```
        if (len != dmxTx.packetSize()) {
        dmxTx.pause();
        while(dmxTx.isTransmitting()) { yield(); }
        dmxTx.setPacketSize(len);
        dmxTx.set(0, msg.data, len);
        dmxTx.resume();
        } else {
        dmxTx.pause();
        while(dmxTx.isTransmitting()) { yield(); }
        dmxTx.set(0, msg.data, len);
        dmxTx.resume();
        }
```

https://user-images.githubusercontent.com/13755765/202873599-404308e2-f432-4329-90b9-62f8d82d53be.mp4

In this situation, the random flickering stops, but with the delay added by this code pushes the writing of data to the buffer to the end of the frame, meaning that you get the end of the frame moving about. This code also causes some flickering, but subjectively, it seems a little bit less.

I then tried using periodic timer (by defining `TEENSY_USE_PERIODICTIMER`), and got this result with the second code:

https://user-images.githubusercontent.com/13755765/202875438-6bdfdbed-333f-4f51-a69d-0d20bdd00fc9.mp4

Unfortunately, I didn't get any data come out for some reason, but I didn't look into it. I note that the beginning of the frame is very stable. Due to all the values being zero, I couldn't tell if the receiving display was glitching.

Finally, I plugged in a stand-alone DMX controller directly into the display, and got the following output:

https://user-images.githubusercontent.com/13755765/202872213-8d164e33-bdfd-45cd-b11d-a19dbc1245ce.mp4

Whilst there is a bit of noise at the low level, you can see that the timing is spot on. This doesn't cause the display to glitch.

I'm wondering if the timing is a little too jittery?

ssilverman, 2022-11-21:
To which fix do you refer when you say “earlier”, and which do you mean by “other”?

Jitter shouldn’t matter. The real question is: “are you seeing invalid DMX packets?” This isn’t a real-time library, so I expect non-uniform DMX frames.

ssilverman, 2023-05-25:
@welshcoder Planning to close this. Any updates on your end?

ssilverman, 2023-05-26:
Thank you for helping to find a couple of fixable issues along the way and for your deep dives here. I’d love to fix all the presented issues at some point, so I’ll come back to this from time to time, if I have a chance. Love to continue working on this with you whenever you have the time.

closed 2023-06-05 by ssilverman