ssilverman/TeensyDMX #16: Not receiveng DMX on teensy 4 with RX monitor on a specific dmx transmitter

DrBit, 2022-08-22:
Hi There!

I've been playing with the code for a long time now and I just encountered a problem that I can not figure a solution yet.
The problem is that I'm unable to receive valid DMX data from one of my DMX transmitters with the teensyDMX library.
I've tried with a hardware node and there I can receive correctly the DMX information coming from the same transmiter.
Clarify that the same code is sucsessfully receiving data coming from a second dmx transmiter from a different brand. So the problem seems to be the format of the DMX data.

I've enabled DMX monitor on pin 6 of the teensy 4,0 but without any luck. Also enabled setKeepShortPackets.

here is my code:

`/*
 * A basic toy receive example, demonstrating single- and
 * multi-byte reads.
 *
 * This example is part of the TeensyDMX library.
 * (c) 2019-2020 Shawn Silverman
 */

#include <cstring>

#include <TeensyDMX.h>

namespace teensydmx = ::qindesign::teensydmx;
elapsedMillis ledTimer;

// Create the DMX receiver on Serial2.
teensydmx::Receiver dmxRx{Serial2};

///////////////////////////////////////////////////////////////////////
// The last value on the channel, for knowing when to print a change
// (Example 1).
uint8_t lastValue = 0;

// Buffer in which to store packet data (Example 2).
uint8_t packetBuf[3]{0};

// The last values received on channels 10-12, initialized to zero.
uint8_t rgb[3]{0};
///////////////////////////////////////////////////////////////////////

void setup() {

  // Turn on the LED, for indicating activity
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWriteFast(LED_BUILTIN, HIGH);

  // initate XBEE serial com
  Serial1.begin (9600);

  // Serial initialization, for printing things
  Serial.begin(115200);
  while (!Serial && millis() < 4000) {
    // Wait for initialization to complete or a time limit
  }
  Serial.println("Starting BasicReceive + send XBEE");

  // Start the receiver
  dmxRx.begin();
  int DMXwachPIN = 6;
  pinMode (DMXwachPIN, INPUT);
  dmxRx.setRXWatchPin (DMXwachPIN);
  dmxRx.setKeepShortPackets(true);

  // Print the first values
  lastValue = dmxRx.get(1);
  // Serial.printf("Channel 1: %d\n", lastValue);

  delay (1000);
}

void loop() {
  // The following two examples print values when they change

  // Example 1. Get the current value of channel 1.
  // This will return zero for no data (and also for data that's zero)
  uint8_t v = dmxRx.get(1);
  if (v != lastValue) {
    lastValue = v;
    Serial1.write (v);    // Send to xbee
    // Serial.printf("Channel 1: %d\n", v);
    setLedAnalog(v);
  }

  if (ledTimer>300) {
    setLedAnalog(0);
  }

  // print any received info from xbee
  if (Serial1.available() > 0) {
    int data = Serial1.read();
    Serial.println (data,HEX);
  }
}

void setLed(bool stateLed) {
  digitalWriteFast(LED_BUILTIN, stateLed);
  ledTimer = 0;
}

void setLedAnalog(uint8_t stateLed) {
  analogWrite(LED_BUILTIN, stateLed);
  ledTimer = 0;
}`

And the scope pictures of the different DMX signals starts (working and not working):

Working signal
[image]

Not working signal
[image]

Any idea of what could be going wrong? Is there anything else I could try that comes to mind?
Thanks for this wonderfull library!

ssilverman, 2022-08-22:
A preliminary note: You don’t need to use the RX watch pin unless you’re trying to measure more accurate Break and MAB values. (Also it must have the DMX input line physically connected to, in your case, pin 6.)

I’m not understanding what the problem you’re seeing is. What do you mean by “unable to receive valid DMX data?” What does this look like and what are you actually seeing?

DrBit, 2022-08-23:
Hi @ssilverman 

Thanks for your answer.

Yes I've got pin 6 connectet directly to RX, in this case I'm using serial 2 of teensy 4.0
What I mean is that even if I'm putting what it seems a valid DMX frame data (received by other DMX node receivers) the code doesn't seem to recognize it as valid.

The code I wrote reads dmx channel 1 and controls the intensity of the builtin led on the teensy. 
when using the signal from one DMX node it works perfectly, but from another it doesn't do anything. It's like it never detects a valid dmx signal.

The captures from the scope are the start frame from both signals, working and non working.
The signal in both are outputing first dmx channel at 255, second at 0 and third at 125, rest all zero.

I've also tried with teensy 3.2 and din't work either.

DrBit, 2022-08-23:
Hi All

Just found a solution to this problem. I have no clue what I did or why it worked but it worked.

In teensyDMX.cpp changed line 12:
from
extern const uint32_t kCharTime    = 11 * kBitTime;         // In microseconds
to
extern const uint32_t kCharTime    = 10 * kBitTime;         // In microseconds

Now the code works with bot DMX nodes and I can receive all data.
I guess is just a timing issue??

Thanks for your support!

ssilverman, 2022-08-23:
Thinking aloud: Maybe the inexactness of the internal timing measurement (this has nothing to do with the RX watch pin) is affecting this? I'm really curious to see a much more detailed measurement of the transmitted DMX from the fixture that doesn't work, specifically an oscilloscope reading of the exact Break and Mark after Break (MAB) times.

Some questions:
1. What exactly is transmitting the non-working DMX stream? Is it a commercial fixture or something custom?
2. Have you tried this same fixture with other DMX receivers? If so, does it work there?

DrBit, 2022-08-23:
To be honest I'm a bit lost when talking about timings :) I know the basics and a bit more but not an expert.

> 1. What exactly is transmitting the non-working DMX stream? Is it a commercial fixture or something custom?

I'm transmitting a test signal on channel 1 , from the grandMA software, a simple wave. And I'm expecting the teensy with teensyDMX to receive the signal.
My test equipment is an original grandMA 2 - 2 ports node and chinese copy of the grandma2 command wing.
When using the node original from MA as output, everything works fine. But when using the output of the chinese commnad wing teensyDMX does not receive any data.

Do you have Logic 2 from seale? I'm sending you a capture of frames from the non working chinese command wing so you can explore all the timings using the logic software. Here the [file](https://filepost.io/d/Sa2r1Xjbmg)
Hope it helps!

> 2. Have you tried this same fixture with other DMX receivers? If so, does it work there?

Yes, I've tried the chinese desk in 4 different theaters, no problems there. Also I tried to loop the output to input of chinese and original MA, and there I could see all the channels being diplayed correctly. It's not extensive research but it worked all the time with all the other fixtures/nodes I've tried.

Thanks!

ssilverman, 2022-08-23:
I have a second question: Where are you measuring that DMX stream, at the MA output or at the Teensy input?

ssilverman, 2022-08-23:
I think it's a system effect of combining the probably-not-exactly 250000 Hz (DMX) baud rate detection of the Teensy coupled with the off-by-1.1-percent output of the knock-off grandMA hardware. Even though it might be less than 2% off, which is within the spec, the system probably sees a > 2% error.

Even the Saleae DMX analyzer is detecting values that are different from what you're intending to send. You're sending zeros for most of the slots, but the detector drifts from reading 0x00 to 0x80 to 0xC0. It's probably trying to align on 4us bit boundaries.

DrBit, 2022-08-23:
You are completely right.

here [capture ](https://filepost.io/d/t7p7T61RPD)of the working ma2 node, all wihtin spec. 

I'm mesuring at the teensy, right at the output of the sn75176. 

Strange that with the chinese stream drifting away and only changing that parameter in the library now all the readings are correct. I've tried last channels and all read ok. So even if it drifts , teensy is reading it correctly with that little change at kCharTime.

By the way, before the change in the library , teensy did not read the packets, but if I did poke around with the conector I tricked it into thinking it was a valid packed and then It did report the right value of the channel I was reading, but only for one frame. The following frames where also not detected or discarted...

ssilverman, 2022-08-23:
So are you saying that changing kCharTime makes it so all slots are correctly read (zeros for most and non-zero for some of the first few)? If so, then it looks like the Teensy serial hardware might be correct, but my checking code could use some fine tuning. I say that because 1.1% off is technically within spec.

ssilverman, 2022-08-24:
I made a new branch with some potential fixes. Can you try this and see if it works for you: https://github.com/ssilverman/TeensyDMX/tree/flexible-char-time

I kept the character time as 44us but reduced it when checking for a bad break.

DrBit, 2022-08-24:
> I made a new branch with some potential fixes. Can you try this and see if it works for you: https://github.com/ssilverman/TeensyDMX/tree/flexible-char-time
> 
> I kept the character time as 44us but reduced it when checking for a bad break.

Your branch works on both chinese and original.
Again checking random channels and all reading correctly.
Let me know if you need further testing.

I've also found another " issue" non important, but just so you know. (maybe is just me that I'm missing something)
I have this in the code in void loop() to check when the DMX line gets disconnected: 

unsigned long tempTimer = millis() - dmxRx.lastPacketTimestamp();
  if (tempTimer >= kTimeout) {
    if (dmx_conected) {
      Serial.println("DMX got disconected");
      Serial.println(tempTimer);
      setLedAnalog(0);
      dmx_conected = false;
    }
  }else{
    if (!dmx_conected) {
      Serial.println("DMX got connected");
      Serial.println(tempTimer);
      dmx_conected = true;
      lastValue = 0;
    }
  }

it work fine almost all the time. tempTimer gives me values betwen 0-8, but once every few minutes tempTimer gets exactly the number 4294967295. Oviously triggers a disconnect briefly in my code that resets when next frame gets in.

Maybe an overflow in dmxRx.lastPacketTimestamp()

Thanks, keep the good work!

DrBit, 2022-08-24:
I'm closing this issue as the original one has been solved. 

I can open a new one for the millis() - dmxRx.lastPacketTimestamp(); if you want

for now I've solved it like this:

`unsigned long tempTimer = millis() - dmxRx.lastPacketTimestamp();
  if (tempTimer >= kTimeout && tempTimer != 4294967295) {
    if (dmx_conected) {
      Serial.println("DMX got disconected");
      Serial.println(tempTimer);
      setLedAnalog(0);
      dmx_conected = false;
    }
  }else{
    if (!dmx_conected) {
      Serial.println("DMX got connected");
      Serial.println(tempTimer);
      dmx_conected = true;
      lastValue = 0;
    }
  }`

closed 2022-08-24 by DrBit

ssilverman, 2022-08-24:
Yes, I have some comments on this. Please open a new issue and I’ll comment there. In the future, you can create new issues when you find new issues. It’s helpful to keep them separate. Thanks for helping me find this one.

PR #17 (2022-08-24): Allow 2% smaller character time when determining a bad break
This fixes a problem where packets with a slightly faster, but within
allowed tolerance, baud rate were being discarded.

Fixes #16