ssilverman/TeensyDMX #19: Receiver is unable to parse RB-DMX1

FloridaManDJ, 2023-01-05:
I'm seeing framing errors and other issues when trying  to receive data from a [Pioneer RB-DMX1](https://www.pioneerdj.com/en-us/product/software/rb-dmx1/dmx-interface/overview/).

My setup is as follows:
- Teensy 4.1
- TeensyDMX was pulled from git, from master branch not more than a few days ago
- Sketch is DMX receiver-only, via `Serial1`. Based on the example sketch.
- I'm using pin 3 as an RX watch pin, via `dmxRx.setRXWatchPin(RX_WATCH_PIN)`
- I'm using the serial console to debug packet stats and error information to the serial monitor.

The RB-DMX1 works fine when I hook it up directly to my lights. I've got roughly 19 DMX devices and about 157 used channels.

The Teensy 4.1 and your TeensyDMX library work fine if I receive DMX data from an Enttec Open DMX, with the same amount of fixtures.

When the RB-DMX1 is sending data to TeensyDMX, I will get framing errors and long packet counts. Depending on what the DMX box is outputting, it will sometimes not give any errors at all, but in other situations, it will give me nothing but errors, with no data at all output. This box is proprietary in that it only works with Rekordbox DJ software. what's weird is that it seems to wig out the most when it's sending certain colors, like red. Other colors/modes, it won't error out at all. Which is really weird.

I started adding debug code in to figure out where the framing errors are coming from. There are about 11 places in the code where `receiveBadBreak()` will increment `framingErrorCount`. 
I see framing error counts increase in the following source locations:
- [Receiver.cpp line 729](https://github.com/ssilverman/TeensyDMX/blob/master/src/Receiver.cpp#L729), space-for-break and MAB was just seen, and either the space-for-break time was too short, or the MAB time was too short
- [Receiver.cpp line 811](https://github.com/ssilverman/TeensyDMX/blob/master/src/Receiver.cpp#L811), First byte is too early, discard any data

The DMX output looks decent on an oscilloscope but my scope kinda sucks. I think I might modify the library to pulse a spare IO pin when the MAB transitions. that would be helpful. But anyway, here are the measurements I can come up with from the scope:

- Space-for-break time: 180.0 uS
- MAB time: ~20 uS
- Data bit length: 4 uS
- Break-to-break time: ~28.8 ms

The times look really consistent, there's not really any jitter to speak of.

I've spent a few days on this and I'm not sure what to do from here. I'm pretty certain that the DMX device is within spec, or close to it, especially since it works with my lights. I think this might a legitimate bug with the library. Could you help me with this?

ssilverman, 2023-01-05:
Thank you for the bug report.

A few questions:
1. What is the MBB time?
2. Did you configure the RX watch pin (3) as an input?
3. May I see some of the scope traces? For getting good scope measurements, I use a Saleae logic analyzer.

Note that `setRXWatchPin()` isn't really needed unless you need to know more accurate BREAK and MAB measurements for, say, testing or statistics or something.

FloridaManDJ, 2023-01-05:
Thanks for the quick reply.

1. MBB time is ~20 microseconds
2. Thankfully I did see that in the docs (very well written docs btw). It's set to an input as the first line of my setup() function
3. Yes, see below, let me know if there are any particular things you want to see, I'll reply with more
4. Yea, it's kind of a mess, and it's accessing error stats that I added to your library for debugging purposes. I guess I'll upload it to github under a fork, and link that here.
5. Yes, it's a custom design but loosely based off of the [Sparkfun Artnet to DMX shield](https://cdn.sparkfun.com/assets/c/1/c/c/d/SparkFun_ESP32_Thing_Plus_DMX_to_LED_Shield.pdf). MAX3485 RS485 converter, and some 6N137's to optically isolate the DMX from the Teensy, and they also act as a level shifter.
6. Yes. Your library works absolutely great in transmit mode by the way. The /RE and DE line on the 485 chip are tied together, they go through a 6N137 to pin 2 on the Teensy. I set this enable line low for receive and high for transmit.

[image]

ssilverman, 2023-01-05:
Some more comments and questions:
1. Could I see a zoomed-in BREAK + MAB + first few bytes?
2. When you send me the trace, I'd like to see what the Teensy sees (pin 0 to ground)
3. Do the BREAK/MAB/bit timings/MBB components always seem correct when looking at a bunch of random packets on the scope?
4. What happens with just the _BasicReceive_ example (with just the minimal modifications to print the channels you're interested in)? Do you still see a failure?
5. If I can't see anything obvious, my next step would be to test it with the device itself.

FloridaManDJ, 2023-01-05:
1. Yep, see below
2. Done, it's measured right from pin 0
3. Yes, they're very consistent
4. That's what I started out with. it would return -1 with no data when it goes into the framing error thing. When I saw the data on the scope I decided to investigate more and then kept looking at more and more metrics.
5. When you say 'test it with the device' which device do you mean? 

[image]

[image]

[image]

[image]

I'm getting the code uploaded to github now. I reinstalled windows 10 recently, I've been lazy and hadn't downloaded git or github desktop since the reinstall. I'll post a link to my fork with the temp library changes and the sketch I'm running as soon as that's done.

ssilverman, 2023-01-05:
A return of -1 is normal because the sampling (`readPacket()`) in `loop()` is much faster than the incoming packet rate. Is it the -1 return values that's concerning to you? (Because you seem to know what you're doing, I'm imagining you know this, but I'm reluctant to leave any stone unturned. :) )

Did you start looking at framing errors because of the -1 return values?

My "test it with the device" comment was referring to my thought that if we can't solve this then I might need to acquire an RB-DMX1 somehow to test with it.

ssilverman, 2023-01-05:
Another approach would be for you to acquire a Saleae device and record several seconds of signal so that I could look at that recorded signal manually.

FloridaManDJ, 2023-01-05:
> A return of -1 is normal because the sampling (readPacket()) in loop() is much faster than the incoming packet rate. Is it the -1 return values that's concerning to you?

I was only concerned because the RB-DMX was belching out data for 10's of seconds, but the sketch was only returning -1, and wouldn't return any data. Or it might return 1 or 2 frames every 5 seconds or so. If it returned -1 but >0 within a few milliseconds then I wouldn't have been concerned and probably wouldn't have caught it. 

> My "test it with the device" comment was referring to my thought that if we can't solve this then I might need to acquire an RB-DMX1 somehow to test with it.

Ah.  It's an expensive box, and they were pretty hard to find at least during the pandemic. I can't part with this one until at least late March, but I suspect I'll have it root-caused before then. I will probably keep banging my head against the wall on this for a few days. I have some hunches but wanted to get the issue open here just in case it was something obvious that I overlooked. If I still can't figure it out then I think I'll probably have to invest in a logic analyzer. It also crossed my mind that I could probably use another Teensy 4.1 to record a few dozen frames to memory and then pipe it all out to the console. At that point I could bring it into a spreadsheet and go over the timings bit by bit.

if it's ok to leave this issue open, I'll update you here when I get it root-caused. Assuming I actually figure it out, I'll be able to send you a pull request.

ssilverman, 2023-01-05:
Thanks for the comprehensive details. Yeah, I don't like to close these until there's some resolution. I appreciate your engagement with this and I'd really love to see this solved. I'll do my best to help.

ssilverman, 2023-01-05:
I wonder how many stop bits are being sent. Are you able to confirm that it's 2 stop bits per byte? I know that's a bit of a tricky ask given the oscilloscope images.

FloridaManDJ, 2023-01-05:
> I wonder how many stop bits are being sent. Are you able to confirm that it's 2 stop bits per byte? I know that's a bit of a tricky ask given the oscilloscope images.

hmm that's a good question. I can figure that out by making a custom fixture sending out a known value of like 0xAA to make it easier to count over to the stop bits. I'll try that either tonight or tomorrow.

> A second request: Can you give me the error printout for, say, 10 seconds of DMX reception? i.e. all the variables in ErrorStats.

yea but there's a [bug in the arduino 2.0 IDE](https://github.com/arduino/arduino-ide/issues/1081) where it doesn't allow you to copy more than 1 screen of data. So that's motivation to uninstall 2.0 and install 1.x. I'm not really liking Arduino 2.0... at all. 

I'll reply back with this info soon... thanks again for the help so far.

FloridaManDJ, 2023-01-05:
[image]

here's a screenshot of the output. 
`framingErrorCount3a` is the `if (mabStartTime_ - breakStartTime_ < kMinBreakTime)` check right after the MAB start and end have just been seen. The `framingErrorCount6` is the `(eopTime - breakStartTime_ < kMinBreakTime + kMinMABTime + charTime)` check where the first byte was too early

I'll send another one with all of the stats enabled.

FloridaManDJ, 2023-01-05:
[image]

With the debug stats

FloridaManDJ, 2023-01-05:
> Would you mind showing me your complete program? 

ok here's the debug sketch that collects metrics. it's a fork of your repo with the framingError metrics-gathering code modified a bit to determine where the framing errors were coming from. My modifications are on the `rb-dmx1` branch.

https://github.com/FloridaManDJ/TeensyDMX/blob/rb-dmx1/examples/DebugMetrics/DebugMetrics.ino

ssilverman, 2023-01-05:
I've attached a test program. Can you try it out and show me some results?
[teensydmx-issue-19.ino.txt](https://github.com/ssilverman/TeensyDMX/files/10356351/teensydmx-issue-19.ino.txt)

FloridaManDJ, 2023-01-06:
[2023-01-05 TeensyDmxIssue19 data.txt](https://github.com/ssilverman/TeensyDMX/files/10356429/2023-01-05.TeensyDmxIssue19.data.txt)

ssilverman, 2023-01-06:
I made a few small changes. Try this:
[teensydmx-issue-19.ino.txt](https://github.com/ssilverman/TeensyDMX/files/10356635/teensydmx-issue-19.ino.txt)

ssilverman, 2023-01-06:
I'm still wondering about stop bit duration. That's the other thing that will trigger a framing error. Also, I'd like to know where `receiveBadBreak()` is being called for you using just my latest test program. (i.e. no RX watch pin or other changes except for enabling RS-485 receive.)

ssilverman, 2023-01-06:
Is it possible to increase the MBB of the packets being sent?

Some more thoughts:
1. It looks like something is getting out of sync and TeensyDMX is having trouble re-synchronizing.
2. In the case of framing errors, that's due to either:
   1. A bad BREAK+MAB: Too short
   2. Stop bits < 8us (i.e. less than 2 stop bits)
3. A BREAK may be detected if there's a framing error and the data was zero, and this includes zero bytes with < 2 stop bits. I wonder if this is what's happening.

FloridaManDJ, 2023-01-06:
> I'm still wondering about stop bit duration. That's the other thing that will trigger a framing error.

I will verify the stop bits now. Will post the results here in a bit.

FloridaManDJ, 2023-01-06:
I measure 20 uS for the stop bits plus mark-between-slots. The start bit is 4 us.

> 1. It looks like something is getting out of sync and TeensyDMX is having trouble re-synchronizing due to the nature of the signals. I'd love to get a view of a longer and cleaner signal.

Agreed but I think I'll need a logic analyzer to do that. The Salae looks nice but I don't think I have $500 to drop on one right now. I could probably use a second Teensy to record a decent bit of raw data to memory and then pipe that out after the capture. If I get that working I'll post the capture here.

> 2. In the case of framing errors, that's due to either:
> A bad BREAK+MAB: Too short

Break+MAB seems to be fine from the scope cature

> Stop bits < 8us (i.e. less than 2 stop bits)

It seems to be consistently 20 us, including the mark between slots.

> I'm noticing that the system is receiving frames at a rate less than 40. Is the RB-DMX1 packet rate configurable, or maybe the low rate is because of so many framing errors?

It is not configurable. Break-to-break seems to be a consistent ~28.8 milliseconds. That's as seen on the scope. That works out to ~34 packets per second. Is that outside the spec? I'm skimming the spec but I can't see where that would be an issue.

ssilverman, 2023-01-06:
So far, everything you’re describing is in-spec. Lower frame rates are indeed in-spec too. I was just asking to increase the MBB because I wonder if a lower frame rate will help things sync better. Certainly a longer inter-slot mark time will do that too. And a 20us inter-slot time is in-spec as well.

I’m stumped. I’m going to see if anyone local has an RB-DMX1.

FloridaManDJ, 2023-01-08:
For the last few days, I've been trying to wrap my head around the receive code. I have a few questions:

1) On [line 125 of LPUARTReceiveHandler.cpp](https://github.com/ssilverman/TeensyDMX/blob/master/src/LPUARTReceiveHandler.cpp#L125), you're checking the `LPUART_STAT_RDRF ` flag to see if the buffer is full. Per the IMXRT reference manual, the flag considers the buffer full when the receive buffer has greater than the number configured into `WATER[RXWATER]`. I don't see anywhere in the code that configures `RXWATER` to some known value, so I assume it's just being set in some part of the Teensy standard library code, like `HardwareSerial.cpp`. If I debug the 32-bit `WATER` register for `Serial1` I get `0x00020002`, meaning that it considers the buffer full when there are > 2 bytes in the FIFO. Doesn't this imply that the receive handler waits for 3 bytes before processing? Wouldn't waiting for 3 bytes make the timestamps less reliable? Especially the first byte, which is of utmost important in the timing measurements, relative to the break and MAB? The code in `irqHandler()` infers the timestamps from previous bytes by subtracting `kCharTime` from the event time, but that only accounts for the start bit, 8 data bits, and 2 stop bits. It doesn't take into account the mark between slots, which in my case is ~12us per byte. I don't know if this causes the issue that I have been seeing, or if it's unrelated, but I'm curious. I'm also curious to see how well the Teensy 4.1 would work in my setup without the FIFO, or with `WATER[RXWATER]` set to 1, either on the first byte, or for the whole packet.

2) In [line 808-809 of Receiver.cpp ](https://github.com/ssilverman/TeensyDMX/blob/master/src/Receiver.cpp#L808-L809), if this code is to ensure the first byte isn't too early, then why not only do this check on the first byte? This is one of the two places that my code is hitting `receiveBadBreak()` and causing the packets to be discarded. I'm not sure why yet but I will be modifying this section a bit to see what happens. Or using more debug code to see how bad it thinks the first byte timing is when it happens compared to normal.

3) When using the RX watch pin, it looks like `mabStartTime_` gets set in `rxPinRose_isr()` and also `receiveIdle()`. I think the latter is only supposed to be happening when the RX watch pin isn't being used. It's inferring the time, but that's not needed when the watch pin is being used, as I understand it. Should there be a missing if block [here](https://github.com/ssilverman/TeensyDMX/blob/master/src/Receiver.cpp#L638-L640)?

ssilverman, 2023-01-10:
I haven't forgotten about this. I'll respond when I next have the bandwidth.

FloridaManDJ, 2023-01-13:
Making these 3 changes appears to have fixed it. I'm going to push the changes to my fork and then try to get the Artnet side of my project working. Then I'll test it for a while with L8CE.

ssilverman, 2023-01-15:
Answers to your questions:
1. Yes, the WATER values are set in _HardwareSerial.cpp_ to 2. And you're right. This line (LPUARTReceiveHandler.cpp:134):
   `uint32_t timestamp = eventTime - kCharTime*avail;`
   assumes there's no inter-slot spacing and absolutely does mess with the timestamp calculation. I need to think about how I want to fix this. One option is as you suggest, to lower the WATER value.
2. You're right again. This calculation is incorrect when there's inter-slot spacing.
3. Could you show me what you mean, specifically?

What were the three changes you made? Thanks for discovering these things. I had, when I was initially writing this, not been considering non-zero received inter-slot times. Thank you for your persistence.

FloridaManDJ, 2023-01-16:
>  One option is as you suggest, to lower the WATER value.

I set it to zero, which means an interrupt will be generated after one byte is received.

> I need to think about how I want to fix this. 

What I did above (setting WATER to zero, forever) was brute-force if anything. I think the best thing to do is to set WATER to zero for the first byte, and then change it back to 2 after that. You don't really care about the timings after the first byte, and the other interrupts will catch frame errors and idle / long break conditions. 

> What were the three changes you made? 

You can find my changes here:
https://github.com/FloridaManDJ/TeensyDMX/tree/rb-dmx1
Look at the 4 commits from January 12th. Ignore the debugging stuff added the week prior to that.

Look the commits over, if they seem benign enough, let me know if you'd like a pull request, or if you have any further questions on them.

> 3. Could you show me what you mean, specifically?

See this commit:
https://github.com/ssilverman/TeensyDMX/commit/3ca806839e98156ad9d045929aba67903d95b47c

And just to reiterate, your library is fantastic overall. The TX side works great and the issues I saw were pretty minor. Overall it's well architected, and that's why I invested the time to get it working with the Pioneer box. Thanks again for all the help.

ssilverman, 2023-01-23:
I've been puzzling over this, and I'd like to try a few things in a sequence of steps. You were correct in identifying some areas of the receiver code that don't take the inter-slot time into consideration, but I've been wondering about a few things.

First, looking at this line:
https://github.com/ssilverman/TeensyDMX/blob/69926c20823763af9c482f8bd578701e60563ad4/src/Receiver.cpp#L809

The variable `eopTime` is the timestamp of the last event. Say an interrupt was just triggered by 3 bytes in the FIFO (WATER is at 2). If there's inter-slot time, then `eopTime` will be later than if there isn't any inter-slot time, with `breakStartTime_` staying the same. Additionally, the variable `charTime`, the total duration of the characters not including any inter-slot time will be smaller than it should be. Because `eopTime` is on the left of the less-than operator and it's bigger, and `charTime` is on the right and it's smaller, not including the inter-slot time will actually push that `if` statement to be false. This means that including the `activeBufIndex_ == 0` check shouldn't make a difference, even though we only intend to check the first byte here.

So my first request is this: Can you see if the problem is fixed _without_ including that `activeBufIndex_ == 0` check from commit bd2377005bd84654e33e7ca70c32d601f915bbb5?

FloridaManDJ, 2023-01-26:
I will hopefully get some time to test these changes this next week. Will let you know what the results are.

ssilverman, 2023-03-17:
I pushed the Receiver change that only checks for a bad break at first byte. See if that helps. Per my logic above, I'm not sure that it would, but I think it's a good change anyway.

See 40f519cc539e5040050206f1d48a05a86a4db9be.

ssilverman, 2023-05-25:
@FloridaManDJ Planning to close this. Any updates on your end?

ssilverman, 2023-06-04:
Closing for now. Please re-open if this is still an issue.

closed 2023-06-04 by ssilverman