ssilverman/TeensyDMX #23: USBProWidget - usb type

netmindz, 2023-05-24:
If I am to understand correctly this example makes the Teensy function like one of the well-known brands DMX Pro interface for use with apps like QLC? How stable is is considered?

I tried to flash, but it's not detected correctly in QLC+ (my other real devices are). Permissions error or am I selecting the wrong USB Type opinion in the IDE?  The file doesn't state the correct mode to use

netmindz, 2023-05-24:
I'm on Linux btw

ssilverman, 2023-05-24:
Does this link provide some insight?
http://www.qlcplus.org/forum/viewtopic.php?t=15183

ssilverman, 2023-05-24:
I just got it to work with _ola_ (macOS). I had to edit my _~/.ola/ola-usbserial.conf_ file to add `cu.usbmodem` as a device prefix line, and then I could see the example running on the Teensy. If your computer can see the serial port then make sure it's visible in that configuration file, and then restart `olad`.

ssilverman, 2023-05-24:
I just tried it in an Ubuntu VM, and the Teensy shows up as `/dev/ttyACM0`, so you can try adding either `ttyACM` or the full `ttyACM0` (meaning it will only find that specific device) to that _ola_ config file and you should see it in the _http://localhost:9090_ page (_olad_ starts the listener there).

ssilverman, 2023-05-25:
In summary: the device-as-USB-pro-widget detection and behaviour works fine with at least _ola_. I also had it working with ENTTEC’s PRO-Manager program a couple years back. I think QLC+ doesn’t work well with most USB serial devices.

stephanschulz, 2023-05-25:
The only change I made was `HardwareSerial &kDMXSerial = Serial2;` instead of `Serial3`
Other than that I use the example as is. :)

netmindz, 2023-05-26:
I did have a quick play with OLA before I posted and I had added the extra prefix, but maybe I made a typo. Let me try that again if that is working for you @ssilverman

netmindz, 2023-05-26:
No ports show on OLA, lsusb shows

Bus 001 Device 027: ID 16c0:0483 Van Ooijen Technische Informatica Teensyduino Serial

Logs  for ola show

plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x1d6b, product 0x0003
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:6 @0x557326c97800 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x0bda, product 0x8152
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:15 @0x557326c97d80 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x0451, product 0xca01
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:16 @0x557326c97880 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x046d, product 0x082b
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:14 @0x557326c97980 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x0451, product 0x8027
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:13 @0x557326c97a00 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x047d, product 0x2048
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:17 @0x557326c97a80 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x413c, product 0x2101
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:12 @0x557326c97b00 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x0451, product 0x8142
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:4 @0x557326c97b80 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x14cd, product 0x1212
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:3 @0x557326c97c00 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x067b, product 0x2586
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:2 @0x557326c97c80 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x1a40, product 0x0801
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 3:1 @0x557326c97d00 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x1d6b, product 0x0002
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 2:1 @0x557326c97e00 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x1d6b, product 0x0003
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 1:4 @0x557326c97e80 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x0c45, product 0x670c
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 1:3 @0x557326c97f00 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x0cf3, product 0xe300
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 1:15 @0x557326c97f80 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x0bda, product 0xc811
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 1:27 @0x557326c9e000 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x16c0, product 0x0483
libs/usb/HotplugAgent.cpp:172: USB hotplug event: 1:1 @0x557326c9e080 [add]
plugins/usbdmx/AsyncPluginImpl.cpp:272: USB device added, checking for widget support, vendor 0x1d6b, product 0x0002
libs/usb/LibUsbThread.cpp:50: -- Starting libusb thread
common/thread/Thread.cpp:200: Thread , policy SCHED_OTHER, priority 0
libs/usb/LibUsbThread.cpp:35: ----libusb event thread is running
olad/PluginManager.cpp:200: Started USB

ssilverman, 2023-05-27:
These are the steps I followed:
1. Start VM running Ubuntu 22.04.1
2. Install _ola_ via `sudo apt install ola`
3. The basic install causes the _olad_ daemon to run at system startup via _init.d_, and it appears to start it after the install
4. If _olad_ is not running, run `olad` at the command line
5. Install latest TeensyDMX from github by copying all the files into ~/Arduino/libraries/TeensyDMX
6. Run the Arduino IDE v2.1.0 with Teensyduino 1.58.1 installed
7. Upload the USBProWidget example
8. Go to http://localhost:9090
9. Expand the "Plugins" list on the left; this seems to populate the _~/.ola/*_ configuration files. If the files aren't populated, try restarting _olad_.
10. Edit _~/.ola/ola-usbserial.conf_ to add `device_prefix = ttyACM` (because the Teensy shows up at _/dev/ttyACM0_)
11. Restart _olad_. I just used the "Stop OLA" button on that OLA home page, and then verified it's not running via the command line, and then just ran `olad`. You may have a different approach.
12. Click the "Add Universe" button on the OLA home page
13. There is an input and an output line that says "Manufacturer Nom - USB Pro Widget (TeensyDMX demo), Serial #: 01020304, firmware 1.44" (those are the temporary names given to the device in _USBProWidget.ino_)

I see this in the _olad_ log:
```
plugins/usbpro/WidgetDetectorThread.cpp:314: Defaulting to a Usb Pro device
plugins/usbpro/WidgetDetectorThread.cpp:330: USB Pro Firmware >= 2.4 is required for RDM support, this widget is running 1.44
```

Hope that helps.

ssilverman, 2023-05-27:
Another detail: Do you have the udev rules installed?
See:
* https://www.pjrc.com/teensy/00-teensy.rules
* (ref: https://www.pjrc.com/teensy/loader_linux.html)

ssilverman, 2023-06-04:
Planning to close this issue. I’ll give it another day.