ssilverman/TeensyDMX #24: using teensy 4.1 for 8 DMX universes

stephanschulz, 2023-05-26:
Does any one have any experience with using the teensy 4.0 or 4.1 as a multi universe DMX interface?
I wonder if it could be used as a replacement for the [ENTTEC Storm 8](chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/https://cdn.enttec.com/pdf/assets/70056/70056_STORM_8_USER_MANUAL.pdf)

I know the Storm 8 is an Art-Net device. So either try to use https://github.com/natcl/Artnet or expand the [USBProWidget](https://github.com/ssilverman/TeensyDMX/tree/master/examples/USBProWidget) example to handle multiple universes. 

I would also need to explore how to expand the ofxDMX openframeworks c++ addon i am currently using so it sends out more than 2 universes.

I am happy to hear if others have done something like this before.

ssilverman, 2023-05-26:
I have a project that converts sACN input to up to 8 DMX outputs, so I know this type of thing works. I use my own sACN stack with https://github.com/ssilverman/QNEthernet and my own HTTP server for configuration.

stephanschulz, 2023-05-26:
I usually try to use off the shelf DMX interfaces like the ENNTEC USB Pro, Storm 8 or Storm 24. 
But I am noticing compared to the Teensy (n a single DMX universe) the Teensy seems to communicate faster to the DMX devices. 
This brought me to this point of wanting to make my own 8 universe device.

ssilverman, 2023-05-27:
I wonder if the ENTTEC devices limit the rate to 40Hz (for full packets). Unless told otherwise, TeensyDMX will send at the maximum rate (for full packets) of about 44Hz.

ssilverman, 2023-05-30:
@stephanschulz were you looking to write one of these yourself or use something already built?

stephanschulz, 2023-05-30:
honestly if someone already has something I would just adopt it. 
I just learned that ENTTEC discontinued their Storm 8 and Storm 24 devices. They have replaced it with a Storm 10.
It's a nice device but maybe having our own USB or Network multi universe device would be better.

ssilverman, 2023-05-30:
I have all the software necessary for the makings of a good device, but no finished hardware to put it on. Want to chat about it off-list?

stephanschulz, 2023-05-30:
Yes please that would be great. Tonight I won't have time but any time starting tomorrow is great. Thank you.

ssilverman, 2023-05-30:
Sure, that works. Send me an email.

ssilverman, 2023-05-30:
@stephanschulz I'm going to close this issue, but may I suggest opening a discussion (should be a link in the same tab bar as "Issues") with the same topic name and initial text?

closed 2023-05-30 by ssilverman